cammill/cammill #11: Many messages arre untraslated/ untraslatable.

onekk, 2015-05-26:
With a quick check of the interface, many items are untraslated (they lacks of the proper msgid in the po file). 

A while ago i inspected the source code and I suspect the issue is in the sourcecode, anyone know how to correct this behaviuor?

TIA and best Regards
carlo D.

koppi, 2015-05-26:
This is easy – here's howto do it: see as an example https://github.com/cammill/cammill/blob/master/main.c#L514

change from
```c
sprintf(tmp_str, "Distance: Mill-XY=%0.2fmm/Z=%0.2fmm / Move-XY=%0.2fmm/Z=%0.2fmm / Time>%0.1fmin", mill_distance_xy, mill_distance_z, move_distance_xy, move_distance_z, milltime);
```
to
```c
sprintf(tmp_str, _("Distance: Mill-XY=%0.2fmm/Z=%0.2fmm / Move-XY=%0.2fmm/Z=%0.2fmm / Time>%0.1fmin"), mill_distance_xy, mill_distance_z, move_distance_xy, move_distance_z, milltime);
```
Add the string ```"Distance: Mill-XY=%0.2fmm/Z=%0.2fmm / Move-XY=%0.2fmm/Z=%0.2fmm / Time>%0.1fmin"``` as a new msgid to you .po file.

For reference, see http://pology.nedohodnik.net/doc/user/en_US/ch-poformat.html
Please test if this works for you.

onekk, 2015-05-26:
Ok i will try, but there is anyone here that could modify the sources, I'm not skilled enough in C and company.
I'm or better was working on a 3d CAM and now with the arrival of the CNC machine, I want to use cammill for 2D and my program for 3D, or better finding someone that can help me on the 3D part, I'm developing it with pyside and python plus the cam library written in C OpenCAMlib.

Thanks

Carlo D.

koppi, 2015-05-26:
I don't have the time right now to do this, sorry.

onekk, 2015-05-29:
Koppi the suggestion you gave me didn't work, if i try what you are suggesting after recompiling cammill the message is unstranslated.

koppi, 2015-05-30:
@onekk push to a branch in your repo and I'll have a look into this.

onekk, 2015-05-31:
ok i will do

onekk, 2015-05-31:
Done see at my branch trouble in the file main.c and it.po (I'm translating some other messages and correcting also some error in the messages in the interface, like cnc-maschine instead of cnc-machine )

koppi, 2015-06-22:
I just created https://crowdin.com/project/cammill

I can be useful to track the progress of the translation and help other uses get involved in translating Cammill to more languages.

closed 2015-07-02 by multigcs