cammill/cammill #14: Can't compile on armhf

onekk, 2015-06-03:
Please I'm trying to compile cammill on a platform with debian wheezy and having as architecture armhf, i installed all the packages with the apt-get line in the readme file, and tryed to compile on this platform.

I get this result:

     clang -ggdb -Wall -Wno-unknown-pragmas -O3 -o cammill main.c pocket.c calc.c hersheyfont.c      postprocessor.c setup.c dxf.c font.c texture.c -lGL -lglut -lGLU -lX11 -lm -lpthread -lstdc++ -lXext -ldl -  lXi -lxcb -lXau -lXdmcp -lgcc -lc `pkg-config gtk+-2.0 --libs` `pkg-config gtk+-2.0 --cflags` `pkg-config gtkglext-x11-1.0 --libs` `pkg-config gtkglext-x11-1.0 --cflags` `pkg-config gtksourceview-2.0 --libs` `pkg-config gtksourceview-2.0 --cflags` `pkg-config lua5.1 --libs` `pkg-config lua5.1 --cflags` -I./ "-  DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED" "-DGSEAL_ENABLE" "-  DHERSHEY_FONTS_DIR=\"./\"" "-DUSE_POSTCAM"
    /tmp/main-TClD7Z.s: Assembler messages:
    /tmp/main-TClD7Z.s:5645: Rd and Rm should be different in mla
    /tmp/main-TClD7Z.s:5748: Rd and Rm should be different in mla
    /tmp/pocket-DvceUF.s: Assembler messages:
    /tmp/pocket-DvceUF.s:128: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s: Assembler messages:
    /tmp/calc-DWq89k.s:1873: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:2063: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:4782: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:5291: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:5693: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:5760: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:5945: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:8164: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:9304: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:10058: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:21785: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:22454: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:22643: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:22874: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:23914: Rd and Rm should be different in mla
    /tmp/calc-DWq89k.s:25312: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s: Assembler messages:
    /tmp/dxf-lbVCx5.s:148: Rd and Rm should be different in mul
    /tmp/dxf-lbVCx5.s:171: Rd and Rm should be different in mul
    /tmp/dxf-lbVCx5.s:188: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:207: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:214: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:223: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:267: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:279: Rd and Rm should be different in mla 
    /tmp/dxf-lbVCx5.s:292: Rd and Rm should be different in mla
    /tmp/dxf-lbVCx5.s:297: Rd and Rm should be different in mla
    mkdir -p intl/de_DE.UTF-8/LC_MESSAGES
    msgfmt po/de.po -o intl/de_DE.UTF-8/LC_MESSAGES/cammill.mo
    mkdir -p intl/it_IT.UTF-8/LC_MESSAGES
    msgfmt po/it.po -o intl/it_IT.UTF-8/LC_MESSAGES/cammill.mo

Cammil  start but all the items in the exander are missing, only some comboboxes are visible.

Feel free to ask some other question, but i haven't the arm machine near the computer and is in a place where is no internet connection. (It's in my CNC control box, near the Mill) 

Regards
Carlo D.

koppi, 2015-06-03:
Try to add:
```
CFLAGS+=-ccc-host-triple armv6-unknown-eabi -march=armv6 -mfpu=vfp -mfloat-abi=hard
```
to the Makefile

koppi, 2015-06-03:
@onekk does this work for you?

onekk, 2015-06-04:
@koppi your code complain a bout the -ccc-host, but here my clang settings (stock debian wheezy on my A20-Olinuxino)

    clang -v
    Debian clang version 3.0-6.2 (tags/RELEASE_30/final) (based on LLVM 3.0)
    Target: arm-unknown-linux-gnueabihf
    Thread model: posix

but here the new error report.
    
    clang -ggdb -Wall -Wno-unknown-pragmas -O3 -o cammill main.c pocket.c calc.c hersheyfont.c     postprocessor.c setup.c dxf.c font.c texture.c -lGL -lglut -lGLU -lX11 -lm -lpthread -lstdc++ -lXext -ldl -lXi -lxcb -lXau -lXdmcp -lgcc -lc `pkg-config gtk+-2.0 --libs` `pkg-config gtk+-2.0 --cflags` `pkg-config gtkglext-x11-1.0 --libs` `pkg-config gtkglext-x11-1.0 --cflags` `pkg-config gtksourceview-2.0 --libs` `pkg-config gtksourceview-2.0 --cflags` `pkg-config lua5.1 --libs` `pkg-config lua5.1 --cflags` -I./ "-DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED" "-DGSEAL_ENABLE" "-DHERSHEY_FONTS_DIR=\"./\"" "-DUSE_POSTCAM" "-march=armv6 -mfpu=vfp -mfloat-abi=hard"
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    /usr/bin/ld: BFD (GNU Binutils for Debian) 2.22 assertion fail ../../bfd/elf32-arm.c:11477
    mkdir -p intl/de_DE.UTF-8/LC_MESSAGES
    msgfmt po/de.po -o intl/de_DE.UTF-8/LC_MESSAGES/cammill.mo
    mkdir -p intl/it_IT.UTF-8/LC_MESSAGES
    msgfmt po/it.po -o intl/it_IT.UTF-8/LC_MESSAGES/cammill.mo

koppi, 2015-06-04:
​Does the executable run?​

onekk, 2015-06-04:
Yes but all the item in the panels are missing only some combobox are showed and the postprocessor description in his place, plus the "graphical" window is flashing with some color.

koppi, 2015-06-04:
Can you make a screenshot?

onekk, 2015-06-04:
I will try ASAP (As i told you my CNC control box is not near my computer and without an internet connection).

koppi, 2015-06-04:
Is it possible to run the debian wheezy arm a20 version in qemu, to
replicate the issue?

onekk, 2015-06-04:
I didn't know but on the site of Olimex there is a guide to compile the kernel on a linux machine, i'm not very skilled in compilator handling, but i'm searching internet about the issue, maybe an issue with clang, even if maybe a qt4 issue due to the flashing problem, my sorce is old and didn0t have the g3d stuff in it but i think, the first step is compiling and visualizing the interface and the solve the display problem, cammill is configured to use the expandable dialog (or wathever they are called) and not the tab like user interface, the "titles" are visible and some comboboxes, no checkboxes or descritption in the interface, menu are visible, maybe later this day i try to make a screenshot, hoping the proper program are installed on the board.

TIA
Carlo D.

onekk, 2015-06-04:
I've found this info using the binutils error as a search key in goggle, maybe it could help?

We did some further investigation on the topic (latest binutils version) and manual builds.

Passing --gc-sections causes the problems on nearly all linker jobs on the ARM platform. Removing that from CCLDFLAGS lets us run the build through.

The issue is known to the binutils team for quite some time: http://sourceware.org/bugzilla/show_bug.cgi?id=14189

Thank you Lennart for giving the right hint.

onekk, 2015-06-04:
The CPU+GPU on the A20-Olinuxino is the same of those on the Cubieboard2

closed 2021-03-26 by koppi