cammill/cammill #15: Request: tidying up the interface.

onekk, 2015-06-05:
Having used cammill for some time now, I've found it useful, but some things are annoying:

1) there is a manner to set a default searching directory for the "Load DXF" dialog
2) it will be useful to have a manner to set the preferred "offset" used in the "Objects" Tab/Expander
3) The "Objects" Tab/Expander have to be higher in the hierarchy, it's one of the most used Tab/Expander so it's annoying to go up and down (or left and right in the Tab view) in the interface.
 A suggested (obviously personal) order could be View - Milling - Objects - Tool - Material - Tabs - Machine - Rotary -Tangential - Misc 
 4) The setting "DXF File " in the view Tab/Expander maybe could be moved int the "Milling" Tab/Expander
5) maybe some items in the Misc like Tolerance could be moved into "Milling" and the others moved to "View"  and "View" renamed "General" (meaning "general settings")

Just my two cents, not mentioning the lack of a proper translation of the few descriptive pages included in doc ("goggle translate" return some funny meaning).

Carlo D.

multigcs, 2015-06-10:
feel free to change the gui, it sounds good to me :)

onekk, 2015-06-14:
@multigcs Hello, i can't figure out some part of your code, I wonder how to put the material settings in the appropriate places, so the "Cutting speed" in the material tab has to set the "feedrate" in the milling tab so the user can tune it if needed or leave the selected rate if it's good.
around line 468 there is a block of code that use the data in the material tab:

    PARAMETER[P_M_FEEDRATE_MAX]

but the calculation have to be put in the appropriate combobox, how to put a value in a combobox?

TIA, thanks four your work  and regards.

Carlo D.

multigcs, 2016-03-06:
4) The setting "DXF File " in the view Tab/Expander maybe could be moved int the "Milling" Tab/Expander

...done

https://github.com/cammill/cammill/commit/63a21eed1f386229317dc502418bfde1ac134561

multigcs, 2016-03-06:
3) The "Objects" Tab/Expander have to be higher in the hierarchy, it's one of the most used Tab/Expander so it's annoying to go up and down (or left and right in the Tab view) in the interface.
A suggested (obviously personal) order could be View - Milling - Objects - Tool - Material - Tabs - Machine - Rotary -Tangential - Misc 

...done

https://github.com/cammill/cammill/commit/5a100ea0b9d93d613dd6cd9afe8119e95662d6f7

closed 2016-09-27 by multigcs