cammill/cammill #16: Tool table: an example file

onekk, 2015-06-09:
The tooltable loading dialog defaults to /tmp dir on Linux. It have to defaults at least to the cammill directory or in the $HOME.

Please anyone could put an example file of a proper tooltable to use as a template to customize one.

Regards

Carlo D.

koppi, 2015-06-09:
For the tool table loading, I changed "/tmp" to "./" which is the current directory cammill is executed from, see: https://github.com/cammill/cammill/commit/1d85e9bb6eb34350f696c3ee0bd6ad59e9aaa316

I added an example tool.tbl file, see: https://github.com/cammill/cammill/blob/1d85e9bb6eb34350f696c3ee0bd6ad59e9aaa316/tool.tbl

koppi, 2015-06-09:
For tool table files, see:

* http://linuxcnc.org/docs/html/gui/tooledit.html
* http://wiki.linuxcnc.org/cgi-bin/wiki.pl?ToolTable

onekk, 2015-06-09:
Many thanks, I'm working on translate the remaining untranslated / missing descriptions on the interface, I'm wondering two things:

1) what are the  "Feedrate/Flute/<4mm" and so on they are present in the material.tbl but i can't find where this values are used in the source to make some calculation or decision.

2) Having a material.tbl in the po or the doc directory maybe with material.tbl.it (fr,en and so on) to have the material name translated.

The main struggle is the lack of a proper documentation, i didn't know german language, and my wife know only the german literature, so I'm using goggle translate to wonder some of the documentation, but the internal are quite difficult to follow, I'm searching to wonder some behaviour looking at the sources, but without a place to discuss it is long and tedious, have anyone thinked about and IRC channel where we "developers" could chat?

Regards

koppi, 2015-06-09:
I just registered #cammill on irc.freenode.org: https://github.com/cammill/cammill/commit/d92a348e5a6b00a3a31f86b099009cd9a1ecfd13

koppi, 2015-06-09:
> 1) what are the "Feedrate/Flute/<4mm" and so on they are present in the material.tbl but i can't find where this values are used in the source to make some calculation or decision.

I guess, that the idea behind this from @multigcs was to make feedrate calculations based on the currently selected material, this is unfinished and there is no cod for this right now.

koppi, 2015-06-09:
> 2) Having a material.tbl in the po or the doc directory maybe with material.tbl.it (fr,en and so on) to have the material name translated.

Just go for id, I'd suggest you add a ```material-it.tbl``` file.

onekk, 2015-06-10:
please but this piece of code in main.c  around line 465 it seem that there is the code to set the PARAMETER[P_M_FEEDRATE_MAX].vint but instead it wil be pre-setting the feedrate in the milling tab.

koppi, 2015-06-10:
@onekk - Don't know what you mean.

onekk, 2015-06-10:
Sorry aorund line 465 in main.c there is this piece of code:

	PARAMETER[P_TOOL_SPEED_MAX].vint = (int)      (((float)Material[PARAMETER[P_MAT_SELECT].vint].vc * 1000.0) / (PI * (PARAMETER[P_TOOL_DIAMETER].vdouble)));
	if ((PARAMETER[P_TOOL_DIAMETER].vdouble) < 4.0) {
		    PARAMETER[P_M_FEEDRATE_MAX].vint = (int)((float)PARAMETER[P_TOOL_SPEED].vint * Material[PARAMETER[P_MAT_SELECT].vint].fz[FZ_FEEDFLUTE4] * (float)PARAMETER[P_TOOL_W].vint);
	} else if ((PARAMETER[P_TOOL_DIAMETER].vdouble) < 8.0) {
		PARAMETER[P_M_FEEDRATE_MAX].vint = (int)((float)PARAMETER[P_TOOL_SPEED].vint * Material[PARAMETER[P_MAT_SELECT].vint].fz[FZ_FEEDFLUTE8] * (float)PARAMETER[P_TOOL_W].vint);
	} else if ((PARAMETER[P_TOOL_DIAMETER].vdouble) < 12.0) {
		PARAMETER[P_M_FEEDRATE_MAX].vint = (int)((float)PARAMETER[P_TOOL_SPEED].vint * Material[PARAMETER[P_MAT_SELECT].vint].fz[FZ_FEEDFLUTE12] * (float)PARAMETER[P_TOOL_W].vint);
	}

koppi, 2015-06-10:
I think, this is unfinished code, maybe @multigcs can fix this?

onekk, 2015-06-10:
Some time ago if I remember well, he wrote it has no time to improve cammill https://github.com/multigcs/cammill/issues/2 

This piece of code didn't seem to update the max-feedrate in the tabs.

closed 2025-02-16 by koppi