cammill/cammill #21: OS X developer wanted

koppi, 2015-06-14:
I did try to fix the OS X build in https://github.com/cammill/cammill/blob/master/.travis.yml
Here's a recent build log on OS X: https://travis-ci.org/cammill/cammill/jobs/66424530
I don't have an OS X machine, maybe someone can have a look into this.

koppi, 2015-06-22:
@multigcs I just ran "make TARGET=OSX osx_app", the resulting dmg image file contains a png from one of your other projects, see screenshot:
[image]

multigcs, 2015-06-22:
right, i copy the generator-script vom my multigcs project, but we must only replace the dmg backgound image.

do you check the osx-app building ? (renamed: make TARGET=OSX package)
i can not test under osx :(

koppi, 2015-06-22:
I don't have a local OS X machine, but travis-ci runs with an OS X node since a few days, see [.travis.yml](https://github.com/cammill/cammill/blob/master/.travis.yml) and build results at:
https://travis-ci.org/cammill/cammill

multigcs, 2015-06-22:
[image]

koppi, 2015-06-22:
"make TARGET=OSX package" passes https://travis-ci.org/cammill/cammill/jobs/67895253

koppi, 2015-06-22:
But:
```bash
cp: utils/cammill.icns: No such file or directory
```
see: https://travis-ci.org/cammill/cammill/jobs/67895253#L2737

multigcs, 2015-06-22:
to genarate the cammill.icns you need 2 commands on osx:
 convert (imagemagick) and png2icns (???)
i can not generate this format under linux :( , but if we have this file, we can put it in git.

see : utils/osx-app.sh (## create/copy icons ##)

koppi, 2015-06-22:
@multigcs https://github.com/cammill/cammill/commit/41561e8a259297a996ac75d69c87a1c1583167bd and https://github.com/cammill/cammill/commit/698b4f549c00e56802f1a9d59f9fba16dcda734f should do it.

phatls, 2015-07-04:
Dear,
Your cammill project is very helpfull. If it's convinient, could you please help give me some guideline for setup enviroment to make plugin function or to custom some function?

Thanks

koppi, 2015-07-04:
@phatls what are you trying to write code for?

phatls, 2015-07-05:
@koppi: Translate into Vietnamese, some thing like that.

koppi, 2015-07-05:
@phatls I just added vi.po to the project, but I ran into another issue, see: #45 hopefully @multigcs has a fix for this.

phatls, 2015-07-06:
@koppi: I want not only Vietnamese but also add function to connect with camera, connect to cloud server, .. If you know, how to setting a project, pls help me.

Thanks

koppi, 2015-07-06:
@phatls Can you describe in more detail, what we can help you with?

closed 2025-02-16 by koppi