cammill/cammill #22: Problem running cammill on FreeBSD in KVM

multigcs, 2015-06-22:
gui will display for a second, but than it crash with follow error-message:

LLVM ERROR: do not know how to split the result of this operator!

koppi, 2015-06-22:
I'm running FreeBSD in qemu, works ok:

```bash
qemu-img create -f qcow2 fbsd.qcow2 20G
qemu-system-x86_64 -m 1G -net nic -net user -s -redir tcp:2022::22 -cdrom ../iso/FreeBSD-10.0-RELEASE-amd64-dvd1.iso fbsd.qcow2
```

multigcs, 2015-06-22:
hmmm, i am using a i386 freebsd-installation, maybe only an issue on 32bit

koppi, 2015-06-22:
On a minimal install with only the xorg package, some icons are missing in the toolbar:
[image]

multigcs, 2015-06-22:
i know, same issue in wine, some stock-images are missing on this systems.
i must check how i can use own images in the toolbar and add this icons to the git.

closed 2025-02-16 by koppi