cammill/cammill #23: Implement / fix Batchmode

koppi, 2015-06-23:
@multigcs do you know how to fix the Batchmode feature?

multigcs, 2015-06-23:
is it broken ? hmm, the only issue i know is, that the gui will popup for a moment

koppi, 2015-06-23:
Oki. batch mode seems to work, but the gui should not popup in batch mode, can you fix this?

multigcs, 2015-06-23:
root@laptop:/usr/src/CadCam/cammill# ./cammill -bm 1 test.dxf  | head -n 20
G21 (Metric)
G40 (No Offsets)
G90 (Absolute-Mode)
 ( GRBL Support only G61 mode )
 ( Version < 0.9i build 20150529 )
 ( must delete the G61 from this file )
 ( otherwise it raise an error )
G61
F100
 (--------------------------------------------------)
 (Object: #0)
 (Layer: 0)
 (Overcut: 0)
 (Depth: -4.000000)
 (Offset: None)
 (--------------------------------------------------)
M05
M6 T1
M03 S10000
G00 X0.0000 Y0.0000 Z4.0000

multigcs, 2015-06-23:
ok, i will try it without gui popup's

multigcs, 2015-06-23:
https://github.com/cammill/cammill/commit/d0d5d91ea9b873a2e9ea114a42d13c476cfa416c

closed 2015-06-23 by multigcs

koppi, 2015-06-23:
This will be very useful, if we want to run some automated tests in the future.

multigcs, 2015-06-23:
we need an gcode validator

multigcs, 2015-06-23:
simple validator:
https://github.com/cammill/cammill/commit/1202148bffefed47717ed8c683b2d9334409570f