cammill/cammill #28: Check and document package building on RPM based Linux distros

koppi, 2015-06-23:
Fedora, SuSE

multigcs, 2015-06-25:
openSUSE done

koppi, 2015-06-25:
@multigcs can you update https://github.com/cammill/cammill/wiki/Linux to reflect the packaging and package installing steps?

multigcs, 2015-06-25:
https://github.com/cammill/cammill/commit/8692afa02def72e326d5ad0ae3fed6c5b1d348f7

koppi, 2015-06-25:
Fedora ftw!

multigcs, 2015-06-25:
centos ?
.........................

koppi, 2015-06-25:
https://getfedora.org/ https://www.centos.org/ CentOS ftw!

multigcs, 2015-06-25:
Fedora installation is running

multigcs, 2015-06-25:
ferdora: wtf, das bisher unsympathischste Linux:
Erlauben sie Standort-Abfrage ?
Wollen Sie Ihre Online-Konten verbinden (FB, Goggle, ...) ?

multigcs, 2015-06-25:
Fedora done 
https://github.com/cammill/cammill/commit/7eb267a5ac8bb433662467cf18768653b8c08cac

multigcs, 2015-06-25:
hmm, centos DVD-ISO is stalled while boot in KVM :(
i must try the netinstall

multigcs, 2015-06-25:
the Minimal-ISO is installed now, but without Xorg (i have select Gnome while install !) *grrrrr

multigcs, 2015-06-25:
after this steps: https://www.centos.org/forums/viewtopic.php?t=47088
i have an gdm login without mouse-curser and than a black window with mouse-curser but no icons, bars or terminals, ctrl-alt-f1 will not work *grrrrrrrrrrrrrr

now, testing live-iso image .......

multigcs, 2015-06-25:
[image]
need more download speed :(

koppi, 2015-06-25:
Here's a list on Wikipedia, if you want em all: https://en.wikipedia.org/wiki/Category:RPM-based_Linux_distributions – If I had a bigger hard drive, I would join the Linux fun, too.

koppi, 2015-06-25:
For the more popular ARM based distros, I would recommend:

* Debian Wheezy / Jessie on ARM, or
* https://www.raspberrypi.org/downloads/
* http://beagleboard.org/latest-images

multigcs, 2015-06-25:
i have a Rasperry-PI but i think its to slow for this application.
i will not support ARM at the moment !

i test arch, gentoo, centos and elementaryos now, i think that is more than 
enough

multigcs, 2015-06-25:
and i will build an simple compile-system like travis for 32/64bit, rpm/deb/tgz/... based on minimal kvm-images.

multigcs, 2015-06-25:
https://github.com/cammill/cammill/issues/34

koppi, 2015-06-25:
As http://www.machinekit.io/ is getting more popular, there are more and more ARM users comming, I would recommend that we have at least one ARM target (preferrably Debian/Jessie ARMhf) for testing.

multigcs, 2015-06-25:
ok, one arm distri (maybe Raspbian) :)

multigcs, 2015-06-25:
ElementaryOS works (based onUbuntu, no changes needed)

multigcs, 2015-06-29:
# find packages/ -type f
packages/FreeBSD/10.0/amd64/cammill_0.9_amd64.tgz
packages/FreeBSD/10.1/i386/cammill_0.9_i386.tgz
packages/CentOS/6.6/i386/cammill_0.9-1_i386.rpm
packages/openSUSE_project/13.2/x86_64/cammill_0.9-1_x86_64.rpm
packages/Debian/8.1/i386/cammill_0.9_i386.deb
packages/Debian/8.1/amd64/cammill_0.9_amd64.deb
packages/Windows/MinGW/IA32/cammill_0.9_IA32.exe
packages/Arch-Linux/3.15.5/x86_64/cammill-0.9-1-x86_64.pkg.tar.xz
packages/Ubuntu/14.04/i386/cammill_0.9_i386.deb
packages/Ubuntu/12.04/i386/cammill_0.9_i386.deb
packages/Fedora/20/i686/cammill_0.9-1_i686.rpm

closed 2015-06-29 by multigcs