cammill/cammill #32: Fix Debian package Lintian checks

koppi, 2015-06-24:
```bash
lintian -i -I --show-overrides  packages/cammill_0.9_amd64.deb
```

```bash
I: cammill: spelling-error-in-binary opt/cammill/cammill Lenght Length
N: 
N:    Lintian found a spelling error in the given binary. Lintian has a list
N:    of common misspellings that it looks for. It does not have a dictionary
N:    like a spelling checker does.
N:    
N:    If the string containing the spelling error is translated with the help
N:    of gettext or a similar tool, please fix the error in the translations
N:    as well as the English text to avoid making the translations fuzzy. With
N:    gettext, for example, this means you should also fix the spelling
N:    mistake in the corresponding msgids in the *.po files.
N:    
N:    You can often find the word in the source code by running:
N:    
N:     grep -rw <word> <source-tree>
N:    
N:    This tag may produce false positives for words that contain non-ASCII
N:    characters due to limitations in strings.
N:    
N:    Severity: minor, Certainty: wild-guess
N:    
N:    Check: binaries, Type: binary, udeb
N: 
E: cammill: unstripped-binary-or-object opt/cammill/cammill
N: 
N:    The package installs an unstripped binary or object file.
N:    
N:    Please note, that shared libraries have to be stripped with the
N:    --strip-unneeded option.
N:    
N:    Refer to Debian Policy Manual section 10.1 (Binaries) and Debian Policy
N:    Manual section 10.2 (Libraries) for details.
N:    
N:    Severity: important, Certainty: certain
N:    
N:    Check: binaries, Type: binary, udeb
N: 
W: cammill: hardening-no-relro opt/cammill/cammill
N: 
N:    This package provides an ELF binary that lacks the "read-only
N:    relocation" link flag. This package was likely not built with the
N:    default Debian compiler flags defined by dpkg-buildflags. If built using
N:    dpkg-buildflags directly, be sure to import LDFLAGS.
N:    
N:    Refer to https://wiki.debian.org/Hardening for details.
N:    
N:    Severity: normal, Certainty: certain
N:    
N:    Check: binaries, Type: binary, udeb
N: 
I: cammill: hardening-no-fortify-functions opt/cammill/cammill
N: 
N:    This package provides an ELF binary that lacks the use of fortified libc
N:    functions. Either there are no potentially unfortified functions called
N:    by any routines, all unfortified calls have already been fully validated
N:    at compile-time, or the package was not built with the default Debian
N:    compiler flags defined by dpkg-buildflags. If built using
N:    dpkg-buildflags directly, be sure to import CPPFLAGS.
N:    
N:    NB: Due to false-positives, Lintian ignores some unprotected functions
N:    (e.g. memcpy).
N:    
N:    Refer to https://wiki.debian.org/Hardening and
N:    http://bugs.debian.org/673112 for details.
N:    
[... 1810 lines cut ...]
N:    avoid making the translations fuzzy. With gettext, for example, this
N:    means you should also fix the spelling mistake in the corresponding
N:    msgids in the *.po files.
N:    
N:    Severity: minor, Certainty: possible
N:    
N:    Check: manpages, Type: binary
N: 
I: cammill: no-md5sums-control-file
N: 
N:    This package does not contain an md5sums control file. This control file
N:    listing the MD5 checksums of the contents of the package is not
N:    required, but if present debsums can use it to verify that no files
N:    shipped with your package have been modified. Providing it is
N:    recommended.
N:    
N:    If you are using debhelper to create your package, just add a call to
N:    dh_md5sums at the end of your binary-indep or binary-arch target, right
N:    before dh_builddeb.
N:    
N:    Severity: wishlist, Certainty: certain
N:    
N:    Check: md5sums, Type: binary
N: 
```

multigcs, 2015-06-24:
we must change the install path for linux and bsd to:
 /usr/bin/cammill, /usr/share/cammill/

all other errors/warnings are fixed

koppi, 2015-06-24:
I still see some other warnings related to file ownership and access control flags:

```bash
$ lintian -i -I --show-overrides  packages/cammill_0.9_amd64.deb 
E: cammill: non-etc-file-marked-as-conffile opt/cammill/material.tbl
N: 
N:    A file installed in some other directory than /etc is marked as
N:    conffile. A conffile typically implies a configuration file, and Policy
N:    mandates such files to be in /etc.
N:    
N:    Refer to Debian Policy Manual section 10.7.2 (Location) for details.
N:    
N:    Severity: serious, Certainty: possible
N:    
N:    Check: conffiles, Type: binary
N: 
E: cammill: non-etc-file-marked-as-conffile opt/cammill/tool.tbl
E: cammill: non-etc-file-marked-as-conffile opt/cammill/postprocessor.lua
E: cammill: non-etc-file-marked-as-conffile opt/cammill/posts/EMC.scpost
E: cammill: non-etc-file-marked-as-conffile opt/cammill/posts/GRBL.scpost
E: cammill: control-file-has-bad-owner conffiles koppi/koppi != root/root
N: 
N:    All control files should be owned by root/root.
N:    
N:    Refer to Debian Policy Manual section 10.9 (Permissions and owners) for
N:    details.
N:    
N:    Severity: serious, Certainty: certain
N:    
N:    Check: control-files, Type: binary, udeb
N: 
E: cammill: control-file-has-bad-owner md5sums koppi/koppi != root/root
W: cammill: description-contains-homepage
N: 
N:    The extended description contains a "Homepage:" pseudo-header following
N:    the old Developer's Reference recommendation. As of 1.14.6, dpkg now
N:    supports Homepage: as a regular field in debian/control. This header
N:    should be moved from the extended description to the fields for the
N:    relevant source or binary packages.
N:    
N:    Severity: normal, Certainty: certain
N:    
N:    Check: description, Type: binary, udeb
N: 
E: cammill: wrong-file-owner-uid-or-gid opt/ 1000/1000
N: 
N:    The user or group ID of the owner of the file is invalid. The owner user
N:    and group IDs must be in the set of globally allocated IDs, because
N:    other IDs are dynamically allocated and might be used for varying
N:    purposes on different systems, or are reserved. The set of the allowed,
N:    globally allocated IDs consists of the ranges 0-99, 64000-64999 and
N:    65534.
N:    
N:    It's possible for a Policy-compliant package to trigger this tag if the
N:    user is created in the preinst maintainer script, but this is a very
N:    rare case and doesn't appear to be necessary. If you found yourself
N:    needing to create a package that works this way, please file a bug
N:    against Lintian to let the maintainers know.
N:    
N:    Refer to Debian Policy Manual section 9.2 (Users and groups) for
N:    details.
N:    
N:    Severity: serious, Certainty: certain
N:    
N:    Check: files, Type: binary, udeb
N: 
W: cammill: non-standard-dir-perm opt/ 0775 != 0755
N: 
N:    The directory has a mode different from 0755, and it's not one of the
N:    known exceptions.
[... 242 lines cut ...]
N: 
W: cammill: non-standard-dir-perm usr/share/doc/cammill/ 0775 != 0755
E: cammill: wrong-file-owner-uid-or-gid usr/share/doc/cammill/README 1000/1000
E: cammill: bad-owner-for-doc-file usr/share/doc/cammill/README koppi/koppi != root/root
W: cammill: non-standard-file-perm usr/share/doc/cammill/README 0664 != 0644
E: cammill: wrong-file-owner-uid-or-gid usr/share/doc/cammill/changelog.Debian.gz 1000/1000
E: cammill: bad-owner-for-doc-file usr/share/doc/cammill/changelog.Debian.gz koppi/koppi != root/root
W: cammill: non-standard-file-perm usr/share/doc/cammill/changelog.Debian.gz 0664 != 0644
E: cammill: wrong-file-owner-uid-or-gid usr/share/doc/cammill/changelog.gz 1000/1000
E: cammill: bad-owner-for-doc-file usr/share/doc/cammill/changelog.gz koppi/koppi != root/root
W: cammill: non-standard-file-perm usr/share/doc/cammill/changelog.gz 0664 != 0644
E: cammill: wrong-file-owner-uid-or-gid usr/share/doc/cammill/copyright 1000/1000
E: cammill: bad-owner-for-doc-file usr/share/doc/cammill/copyright koppi/koppi != root/root
W: cammill: non-standard-file-perm usr/share/doc/cammill/copyright 0664 != 0644
E: cammill: wrong-file-owner-uid-or-gid usr/share/man/ 1000/1000
W: cammill: non-standard-dir-perm usr/share/man/ 0775 != 0755
E: cammill: wrong-file-owner-uid-or-gid usr/share/man/man1/ 1000/1000
W: cammill: non-standard-dir-perm usr/share/man/man1/ 0775 != 0755
E: cammill: wrong-file-owner-uid-or-gid usr/share/man/man1/cammill.1.gz 1000/1000
W: cammill: non-standard-file-perm usr/share/man/man1/cammill.1.gz 0664 != 0644
E: cammill: wrong-file-owner-uid-or-gid usr/share/pixmaps/ 1000/1000
W: cammill: non-standard-dir-perm usr/share/pixmaps/ 0775 != 0755
E: cammill: wrong-file-owner-uid-or-gid usr/share/pixmaps/cammill.png 1000/1000
W: cammill: non-standard-file-perm usr/share/pixmaps/cammill.png 0664 != 0644
```

multigcs, 2015-06-24:
i moved install path to /usr/lib/debian

##
## packages/cammill_0.9_amd64.deb
##
warning: the authors of lintian do not recommend running it with root privileges!
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_128.png
N: 
N:    This package installs a pixmap or a bitmap within /usr/lib. According to
N:    the Filesystem Hierarchy Standard, architecture-independent files need
N:    to be placed within /usr/share instead.
N:    
N:    Severity: normal, Certainty: certain
N:    
N:    Check: files, Type: binary, udeb
N: 
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_16.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_256.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_32.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_48.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_512.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/logo-top.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/logo.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/cfk.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/foam.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/gfk.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/metal.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/plast.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/wood.bmp

koppi, 2015-06-24:
I'm getting
```bash
chmod -R -s packages/debian/ -R
chown -R root:root packages/debian/DEBIAN/
chown: changing ownership of ‘packages/debian/DEBIAN/control’: Operation not permitted
chown: changing ownership of ‘packages/debian/DEBIAN/md5sums’: Operation not permitted
chown: changing ownership of ‘packages/debian/DEBIAN/’: Operation not permitted
Makefile:237: recipe for target 'package' failed
make: *** [package] Error 1
```

koppi, 2015-06-24:
I'm using the powers of sudo:
```bash
sudo make package
```
this works for now.

multigcs, 2015-06-24:
you must do make package as root

please check the last commit, on debian i can not see your error

koppi, 2015-06-24:
```bash
$ sudo make package
strip --remove-section=.comment --remove-section=.note cammill
rm -rf packages/debian
mkdir -p packages/debian/usr/lib/cammill
cp -p cammill packages/debian/usr/lib/cammill/cammill
chmod 755 packages/debian/usr/lib/cammill/cammill
mkdir -p packages/debian/usr/lib/cammill/posts
cp -p posts/* packages/debian/usr/lib/cammill/posts
mkdir -p packages/debian/usr/lib/cammill/textures
cp -p textures/* packages/debian/usr/lib/cammill/textures
mkdir -p packages/debian/usr/lib/cammill/icons
cp -p icons/* packages/debian/usr/lib/cammill/icons
mkdir -p packages/debian/usr/lib/cammill/fonts
cp -p fonts/* packages/debian/usr/lib/cammill/fonts
cp -p material.tbl postprocessor.lua tool.tbl cammill.dxf test.dxf test-minimal.dxf packages/debian/usr/lib/cammill/
mkdir -p packages/debian/usr/bin
ln -sf ../lib/cammill/cammill packages/debian/usr/bin/cammill
mkdir -p packages/debian/usr/share/man/man1/
help2man ./cammill -n "2D CAM-Tool (DXF to GCODE)" | gzip -n -9 > packages/debian/usr/share/man/man1/cammill.1.gz
mkdir -p packages/debian/usr/share/doc/cammill/
cp -p README.md packages/debian/usr/share/doc/cammill/README
echo "It was downloaded from https://github.com/cammill" > packages/debian/usr/share/doc/cammill/copyright
echo "" >> packages/debian/usr/share/doc/cammill/copyright
echo "Copyright 2014 - 2015 by Oliver Dippel <oliver@multixmedia.org>" >> packages/debian/usr/share/doc/cammill/copyright
echo "Copyright 2014 - 2015 by McUles mcules@fpv-club.de" >> packages/debian/usr/share/doc/cammill/copyright
echo "Copyright 2014 - 2015 by Jakob Flierl <@koppi>" >> packages/debian/usr/share/doc/cammill/copyright
echo "Copyright 2014 - 2015 by Carlo <onekk>" >> packages/debian/usr/share/doc/cammill/copyright
echo "License:" >> packages/debian/usr/share/doc/cammill/copyright
echo "" >> packages/debian/usr/share/doc/cammill/copyright
echo "This program is free software; you can redistribute it and/or modify it" >> packages/debian/usr/share/doc/cammill/copyright
echo "under the terms of the GNU General Public License as published by the" >> packages/debian/usr/share/doc/cammill/copyright
echo "Free Software Foundation; either version 2, or (at your option) any" >> packages/debian/usr/share/doc/cammill/copyright
echo "later version." >> packages/debian/usr/share/doc/cammill/copyright
echo "" >> packages/debian/usr/share/doc/cammill/copyright
echo "On Debian systems, the complete text of the GNU General Public License" >> packages/debian/usr/share/doc/cammill/copyright
echo "can be found in the file /usr/share/common-licenses/GPL-3" >> packages/debian/usr/share/doc/cammill/copyright
echo "" >> packages/debian/usr/share/doc/cammill/copyright
git log | gzip -n -9 > packages/debian/usr/share/doc/cammill/changelog.gz
echo "cammill (0.9) unstable; urgency=low\n\n  * Git Release.\n  * take a look in to changelog.gz\n\n -- Oliver Dippel <oliver@multixmedia.org>  `date -R`\n" | gzip -n -9 > packages/debian/usr/share/doc/cammill/changelog.Debian.gz
mkdir -p packages/debian/usr/share/applications
echo "[Desktop Entry]" > packages/debian/usr/share/applications/cammill.desktop
echo "Version=0.9" >> packages/debian/usr/share/applications/cammill.desktop
echo "Type=Application" >> packages/debian/usr/share/applications/cammill.desktop
echo "Name=CAMmill" >> packages/debian/usr/share/applications/cammill.desktop
echo "Comment=2D CAM-Tool (DXF to GCODE)" >> packages/debian/usr/share/applications/cammill.desktop
echo "TryExec=cammill" >> packages/debian/usr/share/applications/cammill.desktop
echo "Exec=cammill %F" >> packages/debian/usr/share/applications/cammill.desktop
echo "Icon=cammill" >> packages/debian/usr/share/applications/cammill.desktop
echo "Categories=Graphics;2DGraphics;Engineering;GTK;" >> packages/debian/usr/share/applications/cammill.desktop
echo "Keywords=cam;cnc;gcode;dxf;" >> packages/debian/usr/share/applications/cammill.desktop
echo "Terminal=false" >> packages/debian/usr/share/applications/cammill.desktop
echo "" >> packages/debian/usr/share/applications/cammill.desktop
mkdir -p packages/debian/usr/share/pixmaps
cp -p icons/icon_128.png packages/debian/usr/share/pixmaps/cammill.png
mkdir -p packages/debian/DEBIAN/
(for F in `find packages/debian -type f | grep -v "^packages/debian/DEBIAN/"`; do md5sum "$F" | sed "s| packages/debian/| |g"; done) >> packages/debian/DEBIAN/md5sums
#(for F in material.tbl tool.tbl postprocessor.lua posts/* ; do echo "/usr/lib/cammill/$F" ; done) >> packages/debian/DEBIAN/conffiles
echo "Package: cammill" > packages/debian/DEBIAN/control
echo "Source: cammill" >> packages/debian/DEBIAN/control
echo "Version: 0.9-`date +%s`" >> packages/debian/DEBIAN/control
echo "Architecture: `dpkg --print-architecture`" >> packages/debian/DEBIAN/control
echo "Maintainer: Oliver Dippel <oliver@multixmedia.org>" >> packages/debian/DEBIAN/control
echo "Depends: libc6, libgtksourceview2.0-0, libgtkglext1, liblua5.1-0" >> packages/debian/DEBIAN/control
echo "Section: graphics" >> packages/debian/DEBIAN/control
echo "Priority: optional" >> packages/debian/DEBIAN/control
echo "Homepage: http://www.multixmedia.org/cammill/" >> packages/debian/DEBIAN/control
echo "Description: 2D CAM-Tool (DXF to GCODE)" >> packages/debian/DEBIAN/control
cat desc.txt | grep ".." | sed "s|^| |g" >> packages/debian/DEBIAN/control
chown -R root:root packages/debian/
[... 87 lines cut ...]
W: cammill: non-standard-file-perm usr/lib/cammill/icons/icon_512.png 0664 != 0644
[... cut ...]
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/metal.bmp
W: cammill: non-standard-file-perm usr/lib/cammill/textures/metal.bmp 0664 != 0644
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/plast.bmp
W: cammill: non-standard-file-perm usr/lib/cammill/textures/plast.bmp 0664 != 0644
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/wood.bmp
W: cammill: non-standard-file-perm usr/lib/cammill/textures/wood.bmp 0664 != 0644
W: cammill: non-standard-file-perm usr/lib/cammill/tool.tbl 0664 != 0644
```

multigcs, 2015-06-24:
https://github.com/cammill/cammill/commit/fd1f9a11aa01d65a5051acf92fa989f3560b2916

koppi, 2015-06-24:
Looks better:

```bash
$ lintian -i -I --show-overrides  packages/cammill_0.9_amd64.deb 
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_128.png
N: 
N:    This package installs a pixmap or a bitmap within /usr/lib. According to
N:    the Filesystem Hierarchy Standard, architecture-independent files need
N:    to be placed within /usr/share instead.
N:    
N:    Severity: normal, Certainty: certain
N:    
N:    Check: files, Type: binary, udeb
N: 
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_16.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_256.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_32.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_48.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/icon_512.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/logo-top.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/icons/logo.png
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/cfk.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/foam.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/gfk.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/metal.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/plast.bmp
W: cammill: image-file-in-usr-lib usr/lib/cammill/textures/wood.bmp
```

koppi, 2015-06-24:
I think, we can also remove the ```icon_???.png``` files.

koppi, 2015-06-24:
I've updated https://github.com/cammill/cammill/wiki/Linux to reflect the need for sudo.

multigcs, 2015-06-24:
hmmm, we must change the code to find the files if we move it to /usr/share/cammill

koppi, 2015-06-24:
Maybe add to the Makefile something like:
```
CFLAGS+=-DPKGDATA=/usr/share/cammill
```

multigcs, 2015-06-24:
hmm, but we need more logic for this,
if we make a test-run (./cammill), we need the data in ./,
after install, we need it in /usr/share/cammill

multigcs, 2015-06-24:
i think thats ok

closed 2015-06-24 by multigcs