cammill/cammill #33: Add a desktop file to the built package

koppi, 2015-06-24:
See: http://standards.freedesktop.org/desktop-entry-spec/latest/apa.html

multigcs, 2015-06-24:
https://github.com/cammill/cammill/commit/c92f13836cb83e39030c8ebc603fe15fb720160c

done

koppi, 2015-06-24:
```bash
lintian -i -I --show-overrides  packages/cammill_0.9_amd64.deb
```
says:
```bash
I: cammill: desktop-entry-lacks-keywords-entry usr/share/applications/cammill.desktop
N: 
N:    This .desktop file does either not contain a "Keywords" entry or it does
N:    not contain any keywords not already present in the "Name" or
N:    "GenericName" entries.
N:    
N:    .desktop files are organized in key/value pairs (similar to .ini files).
N:    "Keywords" is the name of the entry/key in the .desktop file containing
N:    keywords relevant for this .desktop file.
N:    
N:    The desktop-file-validate tool in the desktop-file-utils package is
N:    useful for checking the syntax of desktop entries.
N:    
N:    Refer to
N:    http://standards.freedesktop.org/desktop-entry-spec/latest/ar01s05.html,
N:    http://bugs.debian.org/693918, and
N:    https://wiki.gnome.org/Initiatives/GnomeGoals/DesktopFileKeywords for
N:    details.
N:    
N:    Severity: wishlist, Certainty: certain
N:    
N:    Check: menu-format, Type: binary
N: 
```

multigcs, 2015-06-24:
https://github.com/cammill/cammill/commit/883ebfa1548b6115ee9c30ffed8ef6fc202888cc

closed 2015-06-24 by multigcs

koppi, 2015-06-24:
https://github.com/cammill/cammill/commit/8e0830fccfcc854ad2950d6c2a12f33903c4905c