cammill/cammill #43: Segmentation fault when opening DXF file

WJHildreth, 2015-06-30:
On version 0.9t17-35, I get a few errors when starting the application.  Here is what is reported in the terminal

joeh@IT-JoeH:~$ cammill
Failed to open file '/home/joeh//../share/cammill/icons/icon_128.png': No such file or directory
postprocessor directory not found: /home/joeh//../lib/cammill/posts
unable to open '/home/joeh//../share/cammill/material.tbl'
Segmentation fault (core dumped)
joeh@IT-JoeH:~$ ^C
joeh@IT-JoeH:~$ 

My system is Ubuntu 14.04.2 64bit.  This is the latest source from the repository and built with the instructions on the Wiki.

If I can attach the sample DXF file I am using to test I will.

Joe

multigcs, 2015-06-30:
please test from source-directore without install:

./bin/cammill share/doc/cammill/examples/test.dxf

i changed the get_executable_path function today and have the same problem,
i will try to fix it

thanks for testing :)

multigcs, 2015-06-30:
or use the real path: /usr/bin/cammill

WJHildreth, 2015-06-30:
Using ./bin/cammill share/doc/cammill/examples/test.dxf works without the seg fault as does running it from /usr/bin/cammill

Joe

multigcs, 2015-06-30:
https://github.com/cammill/cammill/commit/a47f65492cc8ce4e82d510c9fd336ad7322be12e

WJHildreth, 2015-06-30:
With this update, (0.9t17-36) I still get the seg fault when running cammill from the commad line but the startup output is a little different.

joeh@IT-JoeH:~$ cammill
cammill - /home/joeh/cammill
cammill - /home/joeh/
Failed to open file '/home/joeh//../share/cammill/icons/icon_128.png': No such file or directory
postprocessor directory not found: /home/joeh//../lib/cammill/posts
unable to open '/home/joeh//../share/cammill/material.tbl'
Segmentation fault (core dumped)
joeh@IT-JoeH:~$ 

Joe

phatls, 2015-07-03:
I aslo meet that mistake:
CucNVs-iMac:cammill-master thodd1$ ./bin/cammill share/doc/cammill/examples/test.dxf
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill/

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed
Failed to open file '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/icons/icon_128.png': Not a directory
postprocessor directory not found: /Users/thodd1/.Trash/cammill-master/bin/cammill//../lib/cammill/posts
unable to open '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/material.tbl'
Segmentation fault: 11
CucNVs-iMac:cammill-master thodd1$

phatls, 2015-07-03:
I have the same problem:
CucNVs-iMac:cammill-master thodd1$ pwd
/Users/thodd1/Downloads/cammill-master
CucNVs-iMac:cammill-master thodd1$ ./bin/cammill share/doc/cammill/examples/test.dxf
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill/

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed
Failed to open file '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/icons/icon_128.png': Not a directory
postprocessor directory not found: /Users/thodd1/.Trash/cammill-master/bin/cammill//../lib/cammill/posts
unable to open '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/material.tbl'
Segmentation fault: 11
CucNVs-iMac:cammill-master thodd1$

phatls, 2015-07-03:
I have the same problem on MacOS, but with linux (Raspberry) it runs ok:
CucNVs-iMac:cammill-master thodd1$ pwd
/Users/thodd1/Downloads/cammill-master
CucNVs-iMac:cammill-master thodd1$ ./bin/cammill share/doc/cammill/examples/test.dxf
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill
./bin/cammill - /Users/thodd1/.Trash/cammill-master/bin/cammill/

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:4702): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed
Failed to open file '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/icons/icon_128.png': Not a directory
postprocessor directory not found: /Users/thodd1/.Trash/cammill-master/bin/cammill//../lib/cammill/posts
unable to open '/Users/thodd1/.Trash/cammill-master/bin/cammill//../share/cammill/material.tbl'
Segmentation fault: 11
CucNVs-iMac:cammill-master thodd1$

phatls, 2015-07-03:
I have the same problem on MacOS, but with linux (Raspberry) it runs ok:
CucNVs-iMac:cammill-master thodd1$ ./bin/cammill share/doc/cammill/examples/test.dxf
./bin/cammill - /Users/thodd1/Downloads/cammill-master/bin/cammill
./bin/cammill - /Users/thodd1/Downloads/cammill-master/bin/cammill/

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed
Failed to open file '/Users/thodd1/Downloads/cammill-master/bin/cammill//../share/cammill/icons/icon_128.png': Not a directory
postprocessor directory not found: /Users/thodd1/Downloads/cammill-master/bin/cammill//../lib/cammill/posts
unable to open '/Users/thodd1/Downloads/cammill-master/bin/cammill//../share/cammill/material.tbl'
Segmentation fault: 11
CucNVs-iMac:cammill-master thodd1$ 
CucNVs-iMac:cammill-master thodd1$

phatls, 2015-07-03:
I have the same problem on MacOS (Yosemite), but with linux (Raspberry) it runs ok:
CucNVs-iMac:cammill-master thodd1$ ./bin/cammill share/doc/cammill/examples/test.dxf
./bin/cammill - /Users/thodd1/Downloads/cammill-master/bin/cammill
./bin/cammill - /Users/thodd1/Downloads/cammill-master/bin/cammill/

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed

(cammill:6067): Gtk-CRITICAL **: gtk_icon_info_get_filename: assertion 'icon_info != NULL' failed
Failed to open file '/Users/thodd1/Downloads/cammill-master/bin/cammill//../share/cammill/icons/icon_128.png': Not a directory
postprocessor directory not found: /Users/thodd1/Downloads/cammill-master/bin/cammill//../lib/cammill/posts
unable to open '/Users/thodd1/Downloads/cammill-master/bin/cammill//../share/cammill/material.tbl'
Segmentation fault: 11

multigcs, 2015-07-03:
https://github.com/cammill/cammill/commit/b888b15f16c75759491ec39d8e0d897c35842137
hope that fix the issue under OSX

multigcs, 2015-07-03:
now batchmode works also in OSX

phatls, 2015-07-06:
Good job :)

koppi, 2015-07-07:
works for me, too.

WJHildreth, 2015-07-16:
version 0.9t17-60 still exhibits the same problem for me.  If I run the program from the desktop launcher and then open a DXF file, the program hangs fro a few seconds then crashes.  If I run cammill from the command prompt from my home directory then open a DXF file it does the same thing with the following text in the terminal.

joeh@IT-JoeH:~$ cammill
cammill - /home/joeh/cammill
cammill - /home/joeh/
Failed to open file '/home/joeh//../share/cammill/icons/icon_128.png': No such file or directory
postprocessor directory not found: /home/joeh//../lib/cammill/posts
unable to open '/home/joeh//../share/cammill/material.tbl'
Segmentation fault (core dumped)
joeh@IT-JoeH:~$ 

If I run it from the source codes bin directory it runs fine.  If I run it from /usr/bin/cammill it runs fine.

I am running Ubuntu 14.04.2 LTS 64 bit.

Do I need to run it on a different platform or machine?

Regards,

Joe

multigcs, 2015-07-16:
have you do a 'make install' again ?
or 'make package' und install the new package ?

WJHildreth, 2015-07-16:
My process was as follows:

git pull
make clean all
sudo make package
double click on the new package and select upgrade through Ubuntu Software Center.

Regards,

Joe

closed 2025-02-16 by koppi