cammill/cammill #44: Unit selection on DXF load

WJHildreth, 2015-07-02:
Is it possible to have a selection available when loading a DXF for either inch or mm?  I have a lot of drawings that are in inches.  When they are loaded, they are displayed as mm.  If using the mm -> inch it converts them from what it thinks is mm to inches.  Or at least that is what it looks like to me.  It makes the drawing so small that you cannot manipulate it.

So for example, if I have a drawing that is 24 inches x 48 inches, the program shows it as mm.  Then when I use the tool, mm -> inches, it converts the 24mm to inches (or roughly 1 inch)

Joe

linuxras, 2016-06-01:
@WJHildreth When I loaded my models the units were in mm but if I just changed the units in Misc tab it updates without using the convert tool at all.

linuxras, 2016-07-16:
Guy please make sure that does not break anything.

closed 2016-09-27 by multigcs