cammill/cammill #45: Makefile: rule "share/locale/%/LC_MESSAGES/cammill.mo: po/%.po" broken"

koppi, 2015-07-05:
@multigcs the Makefile rule "share/locale/%/LC_MESSAGES/cammill.mo: po/%.po" broken" is somehow broken. I just wanted to add new translations, but the mo files do not get updated.

multigcs, 2015-07-06:
the .po file must be newer than the .mo file:

# touch po/*.po
root@station:/usr/src/CADCAM/cammill# make lang
format lang de.po
format lang el.po
format lang fr.po
format lang it.po
format lang ru.po
format lang vi.po
root@station:/usr/src/CADCAM/cammill# ls -l share/locale/*/LC_MESSAGES/cammill.mo 
-rw-r--r-- 1 root root 10830 Jul  6 10:47 share/locale/de/LC_MESSAGES/cammill.mo
-rw-r--r-- 1 root root 13890 Jul  6 10:47 share/locale/el/LC_MESSAGES/cammill.mo
-rw-r--r-- 1 root root 10052 Jul  6 10:47 share/locale/fr/LC_MESSAGES/cammill.mo
-rw-r--r-- 1 root root 10221 Jul  6 10:47 share/locale/it/LC_MESSAGES/cammill.mo
-rw-r--r-- 1 root root 11647 Jul  6 10:47 share/locale/ru/LC_MESSAGES/cammill.mo
-rw-r--r-- 1 root root 11148 Jul  6 10:47 share/locale/vi/LC_MESSAGES/cammill.mo

koppi, 2015-07-06:
@multigcs worked for me. The added po files were fixed in https://github.com/cammill/cammill/commit/e9e49ce3e6a9c76919832f0e1b213802d68dd7c4

closed 2015-07-06 by koppi