cammill/cammill #49: Can't find the procedure to change the language

deusyss, 2015-09-18:
Hi,

I'm the french traductor of Cammill. in the first middle of 2015 year, to change lang, we had just to change the cammill.po before compilation.

I see this week, all the good job done since. But i can't find the way to choose the language. Could you explain me please? I saw tonight that french is no complete with the last version.

Thanks

Best regards

koppi, 2015-09-18:
@deusyss which OS are you running?

deusyss, 2015-09-18:
Hi, thanks for your quick answer. I'm running Debian like (Linux Mint 17).

koppi, 2015-09-18:
Does the following from the commandline
```bash
$ LANGUAGE=fr cammill
```
show something like this?
[image]

deusyss, 2015-09-19:
Hi, thanks for this. I tried same thing, but it works only if i previous build a package, install it on my PC.
It's OK. Thank you very much

closed 2015-09-19 by deusyss

Xav83130, 2016-07-23:
Hi,
Same question for Lubuntu ?
Thx