cammill/cammill #51: Origo is moved when I load an DXF-file

gntech, 2016-02-28:
I have a dxf drawing where I have the origo in the center of the piece, this make sense since the part is symmetrical. However when I load the dxf file it seems like cammill is moving the origo to the corner of the part. This is not good in my case since I dont know where to zero my machine. I need to be able to keep the origo I define in my drawing. Is this possible?

multigcs, 2016-02-29:
Hi,
please use the newest git-version:
https://github.com/cammill/cammill/commit/57afe581f543264505930a07975b21c3b5caf409

You can now setup the zero-point in Misc->Zero (bottom-left/center/original)

I will fix the preview later.

multigcs, 2016-02-29:
Update, much better handling, no reload after changing zero-position:

https://github.com/cammill/cammill/commit/4c4de6745a4882707cdcc05a63b50daaa5f9c970

gntech, 2016-02-29:
Thanks for the quick response! It works perfectly now! :+1:

closed 2016-02-29 by gntech