cammill/cammill #52: No UI updates while mousewheeling

thatch, 2016-05-10:
I inadvertently notice this with one of the holding tab spacing boxes selected... it appears to buffer all the mousewheel events, and the UI freezes until it calculates the last tab position.  Perhapswe could defer this to another thread, or skip intermediate values if the work is going to be thrown away?

Version: current git, f49682

koppi, 2016-05-10:
Hi @thatch , which OS are you running CAMmill on? – My laptop does not have a mousewheel, I cannot reproduce this with my setup. Maybe @multigcs has an idea then?

koppi, 2016-05-10:
@thatch If you already had a look into the source code and know, howto fix this issue, please send a pull request as I'm very busy with other projects right now.

thatch, 2016-05-12:
This was on Linux.  (And actually on a laptop with two-finger scrolling enabled.)

I've already been looking at the source to see about adding other features, just wanted to file this for tracking.

closed 2025-02-16 by koppi