cammill/cammill #56: Arcs results nan on output file

linuxras, 2016-06-01:
I've tried to process a DXF files and found the application to be very intuitive to use, G02 contains I and J code with nan(not a number) references in it like below?

G02 X-5.403946 Y-2.302729 I-nan J-nan

This is not happening to the G03 code.

Any idea what is causing this?

koppi, 2016-06-01:
It's a bug that needs to be fixed. @linuxras can you pastebin a minimal dxf file so that we can reproduce this?

linuxras, 2016-06-01:
Here is the DXF file I am using, nothing propriety so no worries.

http://pastebin.com/usHdQP9S

Thanks for looking into this

koppi, 2016-06-01:
@linuxras can I add your DXF file to the cammill repo as a test case?

koppi, 2016-06-01:
@multigcs Do you have an Idea, what causes this:

```bash
$ bin/cammill -bm 1 share/doc/cammill/examples/test-arc-inch.dxf|grep nan
bin/cammill - /home/koppi/Dropbox/bin/cammill/bin/cammill
bin/cammill - /home/koppi/Dropbox/bin/cammill/bin/
arc({x: -nan, y: -nan, z: 0, angle: -nan});
arc({x: -nan, y: -nan, z: 0, angle: -nan});
arc({x: -nan, y: -nan, z: 0, angle: -nan});
arc({x: -nan, y: -nan, z: 0, angle: -nan});
arc({x: -nan, y: -nan, z: 0, angle: -nan});
```

linuxras, 2016-06-01:
@koppi sure go ahead

koppi, 2016-06-01:
@linuxras ok, thank's – waiting of feedback from @multigcs .

linuxras, 2016-06-05:
@multigcs Just so you know i'm willing to help fix this problem if I can even get some idea where to look.
I've never worked with processing dxf before I have a couple of hours on the weekends between CNC cuts.

koppi, 2016-06-05:
CAMmill has some trouble interpreting object 16, you can see this in the following screenshot on the left side of the cut:
[image]

linuxras, 2016-06-05:
Nut sure what that's about but I removed the round and still get nan

linuxras, 2016-06-07:
It looks like there is definately a problem with that file, I fired up draftsight over the weekend and made up a new design (much simpler) and cammill worked just fine with that.

The only thing I noticed was the inability to put tabs on circles for some reason, all else was fine.

koppi, 2016-07-13:
In https://github.com/cammill/cammill/commit/0478a9af58ed78a91051cdbfc8a657c5465f9b7e @multigcs just started to work on the LWPOLYLINE dxf feature, see screenshot:

[image]

greyltc, 2016-11-13:
I've tried using the latest git and the latest release of this project and they're both (silently) generating NaNs in my output file. I have arcs in my DXFs so I assume it's because of this. This makes this project useless right? or is there some workaround?

Please somehow change cammill to throw a very obvious error when it puts a NaN in the gcode. It can seriously waste a user's time when NaNs make it into the output file with no warning or indication that something has gone terribly wrong.

multigcs, 2016-11-13:
https://github.com/cammill/cammill/commit/dc8d2178973738c80d5523b07e6fe97b8781a83e

greyltc, 2016-11-13:
Super duper!

closed 2025-02-16 by koppi