cammill/cammill #60: Spiral milling

GreatDanton, 2016-09-03:
When milling along the path at several consecutive z-levels, the tool is always dived-in vertically between the levels. This cause some unwanted vibrations. 
Instead of "diving-in" the tool at the beginning of each level,  the tool could travel along the path in a spiral with linearly increasing the Z from Zmin at beginning of the path to the Zmax  at the end of the path at each level.
Is it possible to add this functionality ?

multigcs, 2016-09-05:
nice idea, we have similar function for circles (helix), but i think we can implement this for all operations

multigcs, 2016-09-19:
proof of concept :)
[image]

and a litle preview of the new "treeview" setup

closed 2016-09-27 by multigcs