cammill/cammill #65: Build error in OSX ( Main C )

Wernerhi, 2018-07-17:
Hello 
on my system OSX 10.11
I try to install Cammill and get this error
##########################################################
make TARGET=OSX clean all
rm -rf src/main.o src/pocket.o src/calc.o src/hersheyfont.o src/postprocessor.o src/setup.o src/dxf.o src/hpgl.o src/font.o src/texture.o src/os-hacks.o src/bmp.o
rm -rf bin/cammill

Info:
	TARGET          OSX
	SYSTEM          Darwin
	MACHINE         x86_64
	RELEASE         10.11.6
	DISTRIBUTION    MacOSX
	BINARY          bin/cammill
	VERSION         0.9t17
	VRELEASE        162
	COMP            clang
	PKGS:           gtk+-2.0 gtkglext-1.0 gtksourceview-2.0 lua
	INSTALL_PATH    OSX/CAMmill.app/Contents/MacOS

compile src/main.c -> src/main.o
/bin/sh: pkg-config: command not found
/bin/sh: pkg-config: command not found
/bin/sh: pkg-config: command not found
/bin/sh: pkg-config: command not found
src/main.c:42:10: fatal error: 'gtk/gtk.h' file not found
#include <gtk/gtk.h>
         ^
1 error generated.
make: *** [src/main.o] Error 1
##############################################################
Hombrew is up to date
and all packages installs

greetings Werner

camonroe, 2019-07-10:
Same issue as above with OS 10.14 mojave

multigcs, 2019-07-11:
i don't have osx to test at the moment, sorry